jancumps/pico_scpi_usbtmc_labtool #47: reply when Service Request is invoked

jancumps, 2023-09-22:


jancumps, 2023-09-23:
a maybe relevant [discussion on NI](https://forums.ni.com/t5/Instrument-Control-GPIB-Serial/USB-Interrupt-IN-endpoint-data/td-p/3284087) on the payload

jancumps, 2023-09-23:
Because lack of knowledge, I park this enhancement.

If restarted: this placeholder function in usbtmc_app.c is invoked if the SCPI lib thinks we need to notify:
`void setControlReply ()`

jancumps, 2023-10-01:
retrying

u77345, 2023-10-07:
In the interim, while 1. our PR to TinyUSB is created and accepted and 2. while PICO_SDK bumped up the updated TinyUSB hash we have a few options:

1. Create a build script, to pull down all the dependencies, including libscpi and pico-sdk, into the project build folder, and apply a patch file over pico-sdk's tinyusb. Patch by `git apply`, which is expected to work on all target platforms. 
2. Use add_custom_target() in cmake, to patch timyusb using gnu patch. Poses a risk of `patch` might not be installed.
3. Duplicate all necessary code to bring the src response function up to usbtmc_app from usbtmc_device in the interim. Requires significant amount of code duplication, to recreate the types not importable from tinyUSB.

jancumps, 2023-10-09:
> In the interim, while 1. our PR to TinyUSB is created and accepted and 2. while PICO_SDK bumped up the updated TinyUSB hash we have a few options:
> 
> 1. Create a build script, to pull down all the dependencies, including libscpi and pico-sdk, into the project build folder, and apply a patch file over pico-sdk's tinyusb. Patch by `git apply`, which is expected to work on all target platforms.
> 2. Use add_custom_target() in cmake, to patch timyusb using gnu patch. Poses a risk of `patch` might not be installed.
> 3. Same as above, and add a GitHub CI build to build the pico binary image, taking away developer desktop dependencies on installed tools. Will be clonable and a reference for pico firmware CI build.
> 4. Duplicate all necessary code to bring the srq response function up to usbtmc_app from usbtmc_device in the interim. Requires significant amount of code duplication, to recreate the types not importable from tinyUSB.

I created a destination pull request for this exercise:  [47-reply-when-service-request-is-invoked](https://github.com/jancumps/pico_scpi_usbtmc_labtool/tree/47-reply-when-service-request-is-invoked)

My preference for promoting a solution to the main branch, is to not touch the PicoSDK / TinyUSB sources. I'(m trying to keep the "get-involved" curve for interested parties low.
While working in this dedicated branch, anything goes. We can change build script, patch, ... and tryout first

1. get a version of the code that also supports service requests - no matter of the impact to the code base or build complexity
2. find a way to incorporate that change in the main, without the complexity. Even if that means we have to duplicate (some) TinyUSB functionality.

u77345, 2023-10-10:
While attempting the "no TinyUSB change" approach, discovered: usbtmc_state is declared static, thus can not be extern'd from usbtmc_app.c

`CFG_TUSB_MEM_SECTION static usbtmc_interface_state_t usbtmc_state`

The two key fields picked up from it, report and ep_int_in, appears to be constant on this particular pico and always initialized to {0, 129}. However, when stubbing those out it does not send the SRQ. The patching approach on the other hand works. 

The developer cost is to run a single `git apply` after cloning the pico-sdk
Sent https://github.com/jancumps/pico_scpi_usbtmc_labtool/pull/66 illustrating both.

jancumps, 2023-10-10:
usbtmc_app.c is our own file. 

> The developer cost is to run a single git apply after cloning the pico-sdk
ok with that  

> Docker build is optional.  
I'm not going to import docker / windows / linux specifics in the project. (even optional :) )

jancumps, 2023-10-10:
... I have some conflicts importing this into the project.
Before I try and pull into #47 , can you try and pull #47 changes in your branch, 
then commit and push your branch again.

I'll try to deal with follow up conflicts then.

jancumps, 2023-10-14:
implemented with #67

closed 2023-10-14 by jancumps