jancumps/pico_scpi_usbtmc_labtool #95: Build failure with Pico 2 using SDK 2.1.0

byzantic, 2024-12-03:
Hi, I was trying to build this for Pico 2, using the latest VS Code `pico` extension that uses the SDK v2.1.0.

The VS Code `pico` extension successfully imported the repo and converted the CMakeLists.txt file to use the extension CMake layout.. although I found I had to go through a clean/build cycle to get everything working.

So that built successfully using the original `pico` board.

I then changed the board type to `pico 2` using the extension commands. This changed the project seemingly, however, when I attempted to build, it failed at the link stage:

```
[build] [106/106 100% :: 1.250] Linking CXX executable pico_scpi_usbtmc_labtool.elf
[build] FAILED: pico_scpi_usbtmc_labtool.elf 
[build] : && /home/rob/.pico-sdk/toolchain/13_3_Rel1/bin/arm-none-eabi-g++ -mcpu=cortex-m33 -mthumb -march=armv8-m.main+fp+dsp -mfloat-abi=softfp -mcmse -Og -g -Wl,--build-id=none -Wl,-Map=pico_scpi_usbtmc_labtool.elf.map --specs=nosys.specs -Wl,--wrap=__ctzdi2 -Wl,--wrap=__aeabi_dadd -Wl,--wrap=__aeabi_ddiv -Wl,--wrap=__aeabi_dmul -Wl,--wrap=__aeabi_drsub -Wl,--wrap=__aeabi_dsub -Wl,--wrap=__aeabi_cdcmpeq -Wl,--wrap=__aeabi_cdrcmple -Wl,--wrap=__aeabi_cdcmple -Wl,--wrap=__aeabi_dcmpeq -Wl,--wrap=__aeabi_dcmplt -Wl,--wrap=__aeabi_dcmple -Wl,--wrap=__aeabi_dcmpge -Wl,--wrap=__aeabi_dcmpgt -Wl,--wrap=__aeabi_dcmpun -Wl,--wrap=__aeabi_i2d -Wl,--wrap=__aeabi_l2d -Wl,--wrap=__aeabi_ui2d -Wl,--wrap=__aeabi_ul2d -Wl,--wrap=__aeabi_d2iz -Wl,--wrap=__aeabi_d2lz -Wl,--wrap=__aeabi_d2uiz -Wl,--wrap=__aeabi_d2ulz -Wl,--wrap=__aeabi_d2f -Wl,--wrap=sqrt -Wl,--wrap=cos -Wl,--wrap=sin -Wl,--wrap=tan -Wl,--wrap=atan2 -Wl,--wrap=exp -Wl,--wrap=log -Wl,--wrap=ldexp -Wl,--wrap=copysign -Wl,--wrap=trunc -Wl,--wrap=floor -Wl,--wrap=ceil -Wl,--wrap=round -Wl,--wrap=sincos -Wl,--wrap=asin -Wl,--wrap=acos -Wl,--wrap=atan -Wl,--wrap=sinh -Wl,--wrap=cosh -Wl,--wrap=tanh -Wl,--wrap=asinh -Wl,--wrap=acosh -Wl,--wrap=atanh -Wl,--wrap=exp2 -Wl,--wrap=log2 -Wl,--wrap=exp10 -Wl,--wrap=log10 -Wl,--wrap=pow -Wl,--wrap=powint -Wl,--wrap=hypot -Wl,--wrap=cbrt -Wl,--wrap=fmod -Wl,--wrap=drem -Wl,--wrap=remainder -Wl,--wrap=remquo -Wl,--wrap=expm1 -Wl,--wrap=log1p -Wl,--wrap=fma -Wl,--wrap=cosf -Wl,--wrap=sinf -Wl,--wrap=tanf -Wl,--wrap=atan2f -Wl,--wrap=expf -Wl,--wrap=logf -Wl,--wrap=ldexpf -Wl,--wrap=copysignf -Wl,--wrap=truncf -Wl,--wrap=floorf -Wl,--wrap=ceilf -Wl,--wrap=roundf -Wl,--wrap=sincosf -Wl,--wrap=asinf -Wl,--wrap=acosf -Wl,--wrap=atanf -Wl,--wrap=sinhf -Wl,--wrap=coshf -Wl,--wrap=tanhf -Wl,--wrap=asinhf -Wl,--wrap=acoshf -Wl,--wrap=atanhf -Wl,--wrap=exp2f -Wl,--wrap=log2f -Wl,--wrap=exp10f -Wl,--wrap=log10f -Wl,--wrap=powf -Wl,--wrap=powintf -Wl,--wrap=hypotf -Wl,--wrap=cbrtf -Wl,--wrap=fmodf -Wl,--wrap=dremf -Wl,--wrap=remainderf -Wl,--wrap=remquof -Wl,--wrap=expm1f -Wl,--wrap=log1pf -Wl,--wrap=fmaf -Wl,--wrap=malloc -Wl,--wrap=calloc -Wl,--wrap=realloc -Wl,--wrap=free -Wl,-L/home/rob/work/develop/pico/pico_scpi_usbtmc_labtool/build -Wl,--script=/home/rob/work/develop/pico/pico_scpi_usbtmc_labtool/memmap_custom.ld -Wl,-z,max-page-size=4096 -Wl,--gc-sections -Wl,--no-warn-rwx-segments -Wl,--wrap=sprintf -Wl,--wrap=snprintf -Wl,--wrap=vsnprintf -Wl,--wrap=printf -Wl,--wrap=vprintf -Wl,--wrap=puts -Wl,--wrap=putchar -Wl,--wrap=getchar CMakeFiles/pico_scpi_usbtmc_labtool.dir/source/main.c.o CMakeFiles/pico_scpi_usbtmc_labtool.dir/source/usb/usb_descriptors.c.o CMakeFiles/pico_scpi_usbtmc_labtool.dir/source/scpi/scpi-def.c.o CMakeFiles/pico_scpi_usbtmc_labtool.dir/source/scpi/registers/registers.c.o CMakeFiles/pico_scpi_usbtmc_labtool.dir/source/gpio/gpio_utils.c.o CMakeFiles/pico_scpi_usbtmc_labtool.dir/source/adc/adc_utils.c.o CMakeFiles/pico_scpi_usbtmc_labtool.dir/source/i2c/i2c_utils.c.o CMakeFiles/pico_scpi_usbtmc_labtool.dir/source/adc16/adc16_utils.c.o CMakeFiles/pico_scpi_usbtmc_labtool.dir/source/pwm/pwm_utils.c.o CMakeFiles/pico_scpi_usbtmc_labtool.dir/home/rob/.pico-sdk/sdk/2.1.0/src/rp2_common/pico_stdlib/stdlib.c.o CMakeFiles/pico_scpi_usbtmc_labtool.dir/home/rob/.pico-sdk/sdk/2.1.0/src/rp2_common/hardware_gpio/gpio.c.o CMakeFiles/pico_scpi_usbtmc_labtool.dir/home/rob/.pico-sdk/sdk/2.1.0/src/rp2350/pico_platform/platform.c.o CMakeFiles/pico_scpi_usbtmc_labtool.dir/home/rob/.pico-sdk/sdk/2.1.0/src/rp2_common/pico_platform_panic/panic.c.o CMakeFiles/pico_scpi_usbtmc_labtool.dir/home/rob/.pico-sdk/sdk/2.1.0/src/common/hardware_claim/claim.c.o CMakeFiles/pico_scpi_usbtmc_labtool.dir/home/rob/.pico-sdk/sdk/2.1.0/src/rp2_common/hardware_sync/sync.c.o CMakeFiles/pico_scpi_usbtmc_labtool.dir/home/rob/.pico-sdk/sdk/2.1.0/src/rp2_common/hardware_sync_spin_lock/sync_spin_lock.c.o CMakeFiles/pico_scpi_usbtmc_labtool.dir/home/rob/.pico-sdk/sdk/2.1.0/src/rp2_common/hardware_irq/irq.c.o CMakeFiles/pico_scpi_usbtmc_labtool.dir/home/rob/.pico-sdk/sdk/2.1.0/src/rp2_common/hardware_irq/irq_handler_chain.S.o CMakeFiles/pico_scpi_usbtmc_labtool.dir/home/rob/.pico-sdk/sdk/2.1.0/src/common/pico_sync/sem.c.o CMakeFiles/pico_scpi_usbtmc_labtool.dir/home/rob/.pico-sdk/sdk/2.1.0/src/common/pico_sync/lock_core.c.o CMakeFiles/pico_scpi_usbtmc_labtool.dir/home/rob/.pico-sdk/sdk/2.1.0/src/common/pico_sync/mutex.c.o CMakeFiles/pico_scpi_usbtmc_labtool.dir/home/rob/.pico-sdk/sdk/2.1.0/src/common/pico_sync/critical_section.c.o CMakeFiles/pico_scpi_usbtmc_labtool.dir/home/rob/.pico-sdk/sdk/2.1.0/src/common/pico_time/time.c.o CMakeFiles/pico_scpi_usbtmc_labtool.dir/home/rob/.pico-sdk/sdk/2.1.0/src/common/pico_time/timeout_helper.c.o CMakeFiles/pico_scpi_usbtmc_labtool.dir/home/rob/.pico-sdk/sdk/2.1.0/src/rp2_common/hardware_timer/timer.c.o CMakeFiles/pico_scpi_usbtmc_labtool.dir/home/rob/.pico-sdk/sdk/2.1.0/src/common/pico_util/datetime.c.o CMakeFiles/pico_scpi_usbtmc_labtool.dir/home/rob/.pico-sdk/sdk/2.1.0/src/common/pico_util/pheap.c.o CMakeFiles/pico_scpi_usbtmc_labtool.dir/home/rob/.pico-sdk/sdk/2.1.0/src/common/pico_util/queue.c.o CMakeFiles/pico_scpi_usbtmc_labtool.dir/home/rob/.pico-sdk/sdk/2.1.0/src/rp2_common/hardware_uart/uart.c.o CMakeFiles/pico_scpi_usbtmc_labtool.dir/home/rob/.pico-sdk/sdk/2.1.0/src/rp2_common/hardware_clocks/clocks.c.o CMakeFiles/pico_scpi_usbtmc_labtool.dir/home/rob/.pico-sdk/sdk/2.1.0/src/rp2_common/hardware_pll/pll.c.o CMakeFiles/pico_scpi_usbtmc_labtool.dir/home/rob/.pico-sdk/sdk/2.1.0/src/rp2_common/hardware_vreg/vreg.c.o CMakeFiles/pico_scpi_usbtmc_labtool.dir/home/rob/.pico-sdk/sdk/2.1.0/src/rp2_common/hardware_watchdog/watchdog.c.o CMakeFiles/pico_scpi_usbtmc_labtool.dir/home/rob/.pico-sdk/sdk/2.1.0/src/rp2_common/hardware_ticks/ticks.c.o CMakeFiles/pico_scpi_usbtmc_labtool.dir/home/rob/.pico-sdk/sdk/2.1.0/src/rp2_common/pico_bootrom/bootrom.c.o CMakeFiles/pico_scpi_usbtmc_labtool.dir/home/rob/.pico-sdk/sdk/2.1.0/src/rp2_common/pico_bootrom/bootrom_lock.c.o CMakeFiles/pico_scpi_usbtmc_labtool.dir/home/rob/.pico-sdk/sdk/2.1.0/src/rp2_common/hardware_boot_lock/boot_lock.c.o CMakeFiles/pico_scpi_usbtmc_labtool.dir/home/rob/.pico-sdk/sdk/2.1.0/src/rp2_common/pico_flash/flash.c.o CMakeFiles/pico_scpi_usbtmc_labtool.dir/home/rob/.pico-sdk/sdk/2.1.0/src/rp2_common/hardware_xosc/xosc.c.o CMakeFiles/pico_scpi_usbtmc_labtool.dir/home/rob/.pico-sdk/sdk/2.1.0/src/rp2_common/hardware_divider/divider.c.o CMakeFiles/pico_scpi_usbtmc_labtool.dir/home/rob/.pico-sdk/sdk/2.1.0/src/rp2_common/pico_runtime/runtime.c.o CMakeFiles/pico_scpi_usbtmc_labtool.dir/home/rob/.pico-sdk/sdk/2.1.0/src/rp2_common/pico_runtime_init/runtime_init.c.o CMakeFiles/pico_scpi_usbtmc_labtool.dir/home/rob/.pico-sdk/sdk/2.1.0/src/rp2_common/pico_runtime_init/runtime_init_clocks.c.o CMakeFiles/pico_scpi_usbtmc_labtool.dir/home/rob/.pico-sdk/sdk/2.1.0/src/rp2_common/pico_runtime_init/runtime_init_stack_guard.c.o CMakeFiles/pico_scpi_usbtmc_labtool.dir/home/rob/.pico-sdk/sdk/2.1.0/src/rp2_common/pico_bit_ops/bit_ops_aeabi.S.o CMakeFiles/pico_scpi_usbtmc_labtool.dir/home/rob/.pico-sdk/sdk/2.1.0/src/rp2_common/pico_divider/divider_compiler.c.o CMakeFiles/pico_scpi_usbtmc_labtool.dir/home/rob/.pico-sdk/sdk/2.1.0/src/rp2_common/pico_double/double_math.c.o CMakeFiles/pico_scpi_usbtmc_labtool.dir/home/rob/.pico-sdk/sdk/2.1.0/src/rp2_common/pico_double/double_aeabi_dcp.S.o CMakeFiles/pico_scpi_usbtmc_labtool.dir/home/rob/.pico-sdk/sdk/2.1.0/src/rp2_common/pico_double/double_fma_dcp.S.o CMakeFiles/pico_scpi_usbtmc_labtool.dir/home/rob/.pico-sdk/sdk/2.1.0/src/rp2_common/pico_double/double_sci_m33.S.o CMakeFiles/pico_scpi_usbtmc_labtool.dir/home/rob/.pico-sdk/sdk/2.1.0/src/rp2_common/pico_double/double_conv_m33.S.o CM [... cut ...]
[build] /home/rob/.pico-sdk/toolchain/13_3_Rel1/bin/../lib/gcc/arm-none-eabi/13.3.1/../../../../arm-none-eabi/bin/ld: Binary info must be in first 256 bytes of the binary
```
Looking through the link command, I can see that there is a custom linker script `memmap_custom.ld`, which I suspect is where the problem lies?

This is specified with the following CMake line:
```
pico_set_linker_script(pico_scpi_usbtmc_labtool ${CMAKE_SOURCE_DIR}/memmap_custom.ld)
```
If I comment out this line, then the executable builds OK.

jancumps, 2024-12-03:
The custom script is there, to reserve an area of flash for calibration data ([described in this blog](https://community.element14.com/products/raspberry-pi/b/blog/posts/raspberry-pico-c-sdk-reserve-a-flash-memory-block-for-persistent-storage)).  
When you remove that line, there is a possible side effect, if your code gets large enough to run into the area reserved for calibration data. The build process would not give a warning.  

If you're not planning to store calibration data in your instrument, there is no risk when using the default loader script.

A permanent solution would be to check what the differences are between the Pico and Pico2 memory layout, and either

- adapt the custom loader script so that it's compatible with both, or
- create a dedicated one for Pico 2, and learn the make script to select the right one based on the selected Pico board.

byzantic, 2024-12-03:
OK, understood. At the moment I'm just having a play with the code, so it's not an issue.

There does seem to be a [`hardware_flash` API](https://www.raspberrypi.com/documentation/pico-sdk/hardware.html#group_hardware_flash) .. but it's not clear whether this automatically reserves any space ..

Ideally there would be some mechanism in the SDK to _extend_ the linker script, rather than having to reproduce the original plus changes .. but I haven't looked into that.

jancumps, 2024-12-03:
Looks like it's tagged for SDK 2.2.0 - if review passed and accepted.

ohault, 2025-03-16:
Is Pico 2 currently supported?

jancumps, 2025-03-23:
I don't have that board, but the code only depends on the SDK. Try it by cloning this repo, and set the board in the CMakeLists.txt

Here is how to do it for a Xiao: 
`set(PICO_BOARD seeed_xiao_rp2040)`

closed 2025-03-23 by jancumps

jancumps, 2025-03-23:
see [this conversation and test](https://github.com/jancumps/pst_rtos/issues/1?notification_referrer_id=NT_kwDOAFJpH7MxNTMzNDY3NDk4ODo1NDAwODYz#event-16945317270).